bigtreetech/PandaKnomi #6: Bambu Lab X1C Printers not updating to Knomi's

dfstarr, 2025-03-12:
Hi,

I have tried 6 Panda Knomi's to a couple of X1C printers and none are updating the print status after the printers have been successfully binded. The Panda Knomi stays on the standby gif the whole time and never changes during the whole print. I know they are binded as the Knomi says looking for printer when I turn off the X1C and then when turned on it goes right to the Standby Gif. I have done a reset and flashed the firmware and still the same outcome. It seems like it's a Knomi firmware problem as they work on my Bambu Lab A1 and A1 mini's. Hopefully we can get this resolved fast as I purchased a whole bunch of them and I'm still waiting on more that was ordered directly from Biqu. 

Thanks,

Dan

bigtreetech, 2025-03-12:
On it.

dfstarr, 2025-03-12:
> On it.

Thank you.

bigtreetech, 2025-03-13:
Hi,
I use the latest released version of firmware `01.08.05.00`,
Panda Knomi can recognize the printer status normally, here are some pictures I took.
* Nozzle heating
[image]
* Heatbed heating
[image]
* Push new filament.
[image]
* etc.

This is indeed a bit strange, And what is your X1C firmware version number? So that I can use the same environment to test again?

dfstarr, 2025-03-14:
The current firmware I'm on is 01.08.02. I have also tried the firmware before that one and still the same results. I have tried multiple Panda Knomi's and it's still stuck on the standby gif. It definitely connects to the X1C, but it seems like the Panda Knomi isn't getting the status to report back from the printer. What else can I try?

dfstarr, 2025-03-14:
I finally got it to work. I downgraded the firmware to 01.07.05 and it worked. I then upgraded to 01.08.00 and made sure that worked and then proceeded to the  01.08.02 firmware. It's weird because yesteday I downgraded to 01.08.00 and it didn't work. It seems like the 01.07.05 firmware was needed for some odd reason.

TUnplugged, 2025-03-15:
I am experiencing the same issue.

dfstarr, 2025-03-15:
I had it working for a couple of hours and now it's doing the same exact thing as before. If I downgrade the firmware and then upgrade it again, it works temporarily and then back to no status updates.

dfstarr, 2025-03-15:
> I am experiencing the same issue.

Can you try downgrading the firmware to 01.07.05 and see of that works temporarily?

TUnplugged, 2025-03-15:
I can't find that version to download. It looks like it got pulled.

dfstarr, 2025-03-15:
> I can't find that version to download. It looks like it got pulled.

If you go on the app, select the option to downgrade the firmware and that version shows up for me.

TUnplugged, 2025-03-15:
I'll try that later tonight. I have a print job going on right now.

dfstarr, 2025-03-15:
It seems like this has something to do with the AMS. As soon as I update the AMS color or unplug the ams, the status starts working right away. I have 4 AMS units hooked up. This is weird.....

dfstarr, 2025-03-15:
> I'll try that later tonight. I have a print job going on right now.

How many AMS units do you have plugged in?

dfstarr, 2025-03-16:
I did some further testing and I think I figured out why it stops giving status updates. It only works with 1 AMS. As soon as I disconnect the 2nd cable for the AMS, it starts reporting the status. Once more then 2 AMS units are hooked up, it stops reporting the status on the Knomi Panda. Hopefully we can get a firmware update to fix this.

TUnplugged, 2025-03-16:
I have two AMS hooked up.

dfstarr, 2025-03-16:
> I have two AMS hooked up.

That is why. Disconnect the cable for the 2nd one and I bet it starts to work?

bigtreetech, 2025-03-16:
OK this is useful. It must adjust the format of one of the MQTT status messages when more AMS units are connected. We will get that tested first thing on Monday and get a fix pushed in the next firmware update.

bigtreetech, 2025-03-16:
We were actually about to push the next firmware update which focuses on preserving antialiasing when changing gif colours but we will probably hold it while we tweak the MQTT parser to handle this.

TUnplugged, 2025-03-16:
Yes, I can confirm that going back to 01.07.05 and then flashing up to the latest release, it works. How long it will work is another story.

bigtreetech, 2025-03-16:
> Yes, I can confirm that going back to 01.07.05 and then flashing up to the latest release, it works. How long it will work is another story.

Is that true regardless of how many AMS units you have connected?

TUnplugged, 2025-03-16:
I only tested last night to see if it would gather information from the printer right after the update, and it worked after just starting and then cancelling a print job. This morning, when I went to print something, it did not get info from the printer. I stopped the job. Removed one of the AMS units and then it started to report the printer status again. So it definitely will not get status if the printer has more than one AMS unit attached.

dfstarr, 2025-03-17:
Support sent me a Beta firmware file to address the AMS problem, however I'm having the exact same problem as before?

bigtreetech, 2025-03-18:
> Support sent me a Beta firmware file to address the AMS problem, however I'm having the exact same problem as before?

That is interesting indeed. We found that the latest firmware on the X1C increases the size of the payload coming from the machine which meant that we needed to increase the size of the buffer within the Panda Knomi to accept that payload. We have tested with several AMS units using the latest firmware and we cannot replicate the fault so we are thinking that it must be something unique to your setup. There are a few variables between setups that can affect the MQTT response such as filament types loaded. To help us to narrow down what it may be we were wondering whether you would be able to help us out by manually connecting to your machine and grabbing the MQTT logs and then posting them so that we can see where the difference may be. We can replay your logs to some development Panda Knomi units and then catch the issue.

If you are able to help us out then follow the steps below:

1. Download and install MQTTX.
2. Under the connections tab, select '+' and then 'New connection'.

[image]
3. Enter the details for your machine into the communications settings. You can keep all of the settings the same as what you see below while only changing the IP and password based on your specific machine. You should be able to get the IP address from the machine touch panel on the networking tab and the password will simply be the access code.

[image]

[image]

[image]

4. At this point, you should be able to establish a direct MQTT connection to your printer on your local network by clicking connect.
5. After you have connected, you need to subscribe to a topic in order to start getting status updates from the machine. Click on the relevant button and then enter the topic that you would like to subscribe to. The topic will be the same as the text below with the text in CAPS replaced with your serial number. You can get your serial number from the Panda Knomi UI or from the machine itself. Getting it from the Panda Knomi UI may be easier because it will allow you to copy/paste.

device/YOURSERIALNUMBER/report

[image]

6. As soon as you subscribe, you should start to get info dumps. Let these arrive for about 5 minutes and then you can disconnect.
7. Export the info dumps in JSON format as shown below.

[image]

[image]

8. Open the exported json file using a text editor and then find and remove the line at the start of the file that says "password: xxxzzzzyyyy",
9. Save the file with the password removed and send it to tech@biqu3d.com with luke@bigtree-tech.com in CC.

Thanks for your help.

TUnplugged, 2025-03-18:
I also received the beta firmware and installed it yesterday. It worked with the second AMS attached, and as of this morning, it is still working for me.

dfstarr, 2025-03-18:
I just did some further testing and the first 3 AMS work as expected. When hooking up the 4th AMS unit, it stops reporting the status until I unplug that unit. I initially did a quick test by unplugging the 2nd cord, which is daisy chained to the additional AMS units. When doing that I was getting the same problem as before. Now I did a more precise test of process of elimination of each unit. So basically 3 AMS units will work with the beta firmware and once the 4th unit is plugged in, it stops reporting. We are close to fixing the problem, just need to fix that 1 last AMS unit.. Did you still want me to do the above steps? Thanks.

bigtreetech, 2025-03-18:
> I just did some further testing and the first 3 AMS work as expected. When hooking up the 4th AMS unit, it stops reporting the status until I unplug that unit. I initially did a quick test by [...]

We would really appreciate it. I know it looks like a lot of steps but it is actually quite quick and easy. It will help us to understand what it is that your fourth AMS is doing that the Panda Knomi is not liking.

dfstarr, 2025-03-18:
> We would really appreciate it. I know it looks like a lot of steps but it is actually quite quick and easy. It will help us to understand what it is that your fourth AMS is doing that the Panda [...]

Not a problem. I will do it sometime today.

dfstarr, 2025-03-18:
> We would really appreciate it. I know it looks like a lot of steps but it is actually quite quick and easy. It will help us to understand what it is that your fourth AMS is doing that the Panda [...]

I just emailed 2 dumps. I did 1 for 3 AMS's connected and 1 for 4 AMS's connected. Wasn't sure if you needed both.

Tenkiller77, 2025-03-19:
I'm having the same issue with my X1C. I used the USB cable and the 2 pin connector and still no statis update from printer. I can log in to the Panda knomi via IP and it show the binding to the printer and I can even change the .gif's. my printer firmware is 01.08.02.00. The problem is I did not want to upgrade the firmware because Bambu's lock down.

dfstarr, 2025-03-19:
> I'm having the same issue with my X1C. I used the USB cable and the 2 pin connector and still no statis update from printer. I can log in to the Panda knomi via IP and it show the binding to the [...]

How many AMS units?

Tenkiller77, 2025-03-19:
I have 2 AMS's

dfstarr, 2025-03-19:
Unless your have the beta firmware, it wont work with 2 ams. Unplug the 2nd one and it will start reporting the status.

Tenkiller77, 2025-03-19:
I will try that, but unfortunately I need both AMS I'm running a business
with them.

dfstarr, 2025-03-19:
> I will try that, but unfortunately I need both AMS I'm running a business
> with them.
> […](#)

I meant just to test it out. I'm sure they will fix it soon.

Tenkiller77, 2025-03-19:
I bit the bullet and upgraded the firmware on the Bambu X1C and it still
doesn't work,  but I still have both AMS attached. It's printing right now
and after it's done I'll remove one and try it again.

dfstarr, 2025-03-19:
> I bit the bullet and upgraded the firmware on the Bambu X1C and it still
> doesn't work,  but I still have both AMS attached. It's printing right now
> and after it's done I'll remove one and try it again.
> […](#)

You have to wait for the Panda Knomi firmware. It should be soon as I tested the beta and it now works with all the AMS's connected.

Tenkiller77, 2025-03-19:
OK I'm testing with 1 AMS and  now it's working. Thanks for your help.

loloboss83, 2025-03-19:
Hello is it possible to share the beta firmware ? because i have 2 ams and i have no information on knomi ?

Thanks

dfstarr, 2025-03-20:
> Hello is it possible to share the beta firmware ? because i have 2 ams and i have no information on knomi ?
> 
> Thanks

I will ask if it's ok to share.

dfstarr, 2025-03-20:
I tested the latest beta irmware amd all 4 AMS's are working.

loloboss83, 2025-03-20:
> > Hello is it possible to share the beta firmware ? because i have 2 ams and i have no information on knomi ?
> > Thanks
> 
> I will ask if it's ok to share.

ok thanks !

bigtreetech, 2025-03-20:
> > Hello is it possible to share the beta firmware ? because i have 2 ams and i have no information on knomi ?
> > Thanks
> 
> I will ask if it's ok to share.

Please hold off on sharing. The problem with this is that beta firmware can have bugs in it and when it is widely shared, people may think that it is final and then they get unhappy with the bugs. We prefer to host dedicated, private github repos for beta releases where those who understand the nature of beta firmware are welcome to join and participate with feedback. We will get this set up but in the meanwhile, @loloboss83 please just email luke@bigtree-tech.com and we will get it sent to you.

Buyco-dotcom, 2025-03-23:
> OK this is useful. It must adjust the format of one of the MQTT status messages when more AMS units are connected. We will get that tested first thing on Monday and get a fix pushed in the next firmware update.

Buyco-dotcom, 2025-03-23:
The AMS problem doesn't just affect the Panda Knomi, but also the Panda Touch. If the Panda Touch receives data via LAN mode and not cloud mode, it only works with a maximum of two AMS units. The Knomi only works with one AMS. With the new security update from Bambulab, both LAN mode and developer mode must be enabled in order to control the Bambu with the Touch. With more than two AMS units, however, operation will not be possible due to the error, and the Touch will become unusable with the update.

bigtreetech, 2025-03-23:
> The AMS problem doesn't just affect the Panda Knomi, but also the Panda Touch. If the Panda Touch receives data via LAN mode and not cloud mode, it only works with a maximum of two AMS units. The Knomi only works with one AMS. With the new security update from Bambulab, both LAN mode and developer mode must be enabled in order to control the Bambu with the Touch. With more than two AMS units, however, operation will not be possible due to the error, and the Touch will become unusable with the update. I already submitted a request to the PandaTouch repository on November 6, 2024. So far, no response.

There will be a patch for the touch this week if there isn't already one included in 1.0.7. Personally, I have not reviewed 1.0.7 as I have been busy with some other projects and the Panda Touch developers have been working on 1.0.7 but I will check in with them tomorrow as to whether or not a patch for this was included in 1.0.7 and if not then we will push a quick 1.0.7.1.

tronibirk, 2025-03-24:
Schade das es noch nicht mit 2Ams fuktioniert das Panda Knomi

Eyekiller2006, 2025-03-29:
> There will be a patch for the touch this week if there isn't already one included in 1.0.7. Personally, I have not reviewed 1.0.7 as I have been busy with some other projects and the Panda Touch [...]

Any idea when this will release?