bigtreetech/PandaKnomi #8: Getting a Printer Error image when A1 starts printing

sdsviet, 2025-03-16:
I recieved my panda knomi today and I got everything installed and it started working. 

It goes thru all the calibration animations no problem. once it gets to the printing, I get a printer error image with a QR code to bambu's x axis is too low error webpage. 

After that it stays on that image. I've restarted the a1, unplug and replug all cables, updated img and firmware bin files and even factory reset the panda knomi. Only a restart would lets you see the calibration animations and then back to the printer error image. 

The only firmware I found was the standard v1.02 bin file from web searches. Are there any troubleshooting steps I can take. I am using an AMS with the A1 as well. 

[image]

bigtreetech, 2025-03-16:
This is a legitimate error that your printer is reporting. Please follow the guidelines on the Bambu wiki linked to by that QR code under the A1 section to tighten your belt. After that, the error should disappear when you start a new print.

sdsviet, 2025-03-16:
Thank you I will check it out. I have the Bambu Studios app open and I’ve
gone thru the touchscreen on the A1 and there’s no notification of it but
I’ll double check again.

sdsviet, 2025-03-16:
Thank you for the advice. It was an issue with the printer itself. I only
saw the notification in the Bambu handy app but not anywhere in Bambu
studios or the printer itself. Everything seems to be working fine now.

bigtreetech, 2025-03-18:
Glad to hear that the Panda Knomi helped to draw your attention to this and get it sorted :)

closed 2025-03-18 by bigtreetech