bigtreetech/PandaKnomi #15: Changed Hostname and Knomi Died

cgtracy, 2025-03-24:
I changed the hostname from the default and the unit rebooted then died. The red power light is on but there is no screen, no Wi-Fi AP, and no local network connection (no IP). Tried to power cycle but will not return to operational status.

bigtreetech, 2025-03-24:
Interesting. Could you share what you changed the hostname to? It may be that we need to add some extra sanity checks to the hostname field (special characters or a string that is too long may be able to cause this). To recover your knomi, please follow the steps at this link. You will need to download the factory firmware to the Knomi using USB-C. https://bttwiki.com/PandaKnomi.html#flash_download_tool-type-c

cgtracy, 2025-03-24:
Of course! It was changed from default PandaKnomi to PandaKnomiFrancois. I'm guessing too long for the usual 15 character standard. I have a second unit that was renamed to a hostname of 14 characters and had no issue.

cgtracy, 2025-03-25:
Unfortunately, the factory firmware instructions did not revive the unit. Of note, the download links in the Wiki to the flash tool comes up with a 403 authentication error. The .com version of the site worked and I was able to download the flash tool.

I followed the instructions precisely but the little guy is just dead now. I will need to arrange for a replacement.

bigtreetech, 2025-03-25:
> Unfortunately, the factory firmware instructions did not revive the unit. Of note, the download links in the Wiki to the flash tool comes up with a 403 authentication error. The .com version of the [...]

The 403 error is really odd. It seems to be erratic. I just tried it after reading your message and it worked. I then decided to double check about 2 minutes later and it threw a 403. Changing it to the com link anyways for reliability.

I have checked with the firmware developers who worked on this project and they have confirmed that there was an issue with the V1.0.2 firmware that would result in the panda knomi being unable to start up if the hostname exceeded 15 characters. This was corrected in V1.0.3 

The problem is that when you reprogrammed the knomi it only reprogramed the firmware part but left the non-volatile storage part (where the hostname is stored) intact. So after it booted with the new firmware, it again read the hostname and that is why it didn't work.

There are two routes to get it going again:

1. Reprogram with the latest beta of 1.0.3 via USB-C (no need to do a full chip erase so non-volatile data is retained).
2. Do a full chip erase before you program the new firmware. The full chip erase is performed using the erase button at the bottom of the ESP tool. This will also erase the WiFi username, password and the printer binding details. It will be as if it shipped from the factory.

Let me know if you want to try the latest beta and I can mail it to you.

bigtreetech, 2025-03-25:
By the way, the latest beta also has a completely new method to change the colour of GIFs which retains anti-aliasing.

cgtracy, 2025-03-25:
It took a few attempts but I was finally able to get it working again using
the erase/reflash method.

I'd be happy to try the beta firmware now that I've got it back up and
running. Any other release notes?

closed 2025-03-31 by bigtreetech