bigtreetech/PandaKnomi #16: Request: Allow for different print statistics to be shown on the Knomi

goldsam328, 2025-03-25:
Currently the Knomi only displays the current print percentage. It would be nice to be able to cycle between other stats like remaining time, current layer/total layers, actual end time, etc.

bigtreetech, 2025-04-09:
Alright, we hear you guys ;)

oakyz, 2025-11-01:
And still no updates or any of the requested features added….

goldsam328, 2025-11-02:
> And still no updates or any of the requested features added….

It's very disappointing but I've already accepted that the company has given up on this product. I practically threw my Knomi in for free with my A1 when I sold it on FB marketplace.

oakyz, 2025-11-02:
> > And still no updates or any of the requested features added….
> 
> It's very disappointing but I've already accepted that the company has given up on this product. I practically threw my Knomi in for free with my A1 when I sold it on FB marketplace.

It`s not like there`s complicated requests either, to be able to show some other statistics while printing apart from a percentage number and a bar..... @bigtreetech in the "pipeline" for months and the new "resources " that are "chipping away" with no results in months...