bigtreetech/PandaKnomi #18: Panda Knomi won't boot

niadhk, 2025-04-04:
I received my Panda Knomi today, & was going through the binding process (I got to the screen where I input my home wifi password), when the screen went black. Now the screen stays black, with the red led behind it, but I cannot reboot or factory reset, pressing the boot button, or long pressing for 5 seconds (I've tried up to 20 seconds) does nothing, screen stays black.
I've tried forcing the firmware update wia USB C, but the green panel on the Flash Download Tool is stuck on SYNC. The black screen has dots populating, but has been doing it for approx 20 minutes, with no visible sign of progress.

chuckkay, 2025-04-05:
Same thing happened to me last month, when i set everything up but when I changed the hostname to something else it rebooted to a black screen and a bricked device. I Reported the bug and they acknowledged it hostname longer the 15 characters kills the device, the had to send me a new one, now they posted 1.3.1 firmware update which i hope fixes the problem

nfrobertson, 2025-05-27:
> Update the firmware to the latest version on this repo (https://github.com/bigtreetech/PandaKnomi/tree/master/Firmware/v1.0.3.1) using this guide:
> 
> https://bttwiki.com/PandaKnomi.html#flash_download_tool-type-c
> 
> There was a bug with the initial firmware where some fields could only be set to 15 chars max. It has been fixed and other improvements have been made in the latest firmware.

Any idea how long this is supposed to take?   I am running the flash download tool and flashing this to 1.0.3.1 after carefully reviewing all the settings.  It' been running a couple hours now which seems too long?  The flash tool console has a constant flow of dots making me think it is still working but I have no idea how long this should run?   

I do see the fix in 1.3 you mention and recall one of the last things I did was sent the network name to a string that would have been 16 characters so over the 15 limit.

bigtreetech, 2025-05-27:
Definitely not a few hours. I would just unplug and try again. The bootloader cannot be damaged so you can't brick it.

bigtreetech, 2025-05-28:
Good stuff. I want to leave it open for a while to make it more visible in case anyone else has the same issue.

ThorinOkenshield, 2025-06-23:
Hi All, i am having the exact same problem. i am trying to update the firmware after my knomi stopped working and the screen went black when powered on. 
when i start the flash tool, all that happens is i get the dots for ever and nothing seems to progress. ive tried plugging it into other usb ports, holding the reset button whilst inserting the usb and whilst it is connected to the pc, but nothing is helping? 

The DetectedInfo section also isn't populating with any information? 

[image]

xxDUFFYxx, 2025-07-02:
Just stop it if you get the dots.
Follow the instructions and I had to download the files again and add them to get it to work, even added an extra image under the 1.031 to it and it worked.

image.png (view on web)<https://github.com/user-attachments/assets/30608b38-f2ff-4e5b-84ef-a28e25b91202>

i have the same Problem, could you solve it in the meantime

ThorinOkenshield, 2025-07-02:
> Just stop it if you get the dots.
> Follow the instructions and I had to download the files again and add them to get it to work, even added an extra image under the 1.031 to it and it worked.
> 
> Sent from Outlook for Android<https://aka.ms/AAb9ysg>
> […](#)

I redownloaded the tool multiple times and tried different usb a/c ports on my pc and the usb port on my Surface.

ThorinOkenshield, 2025-07-03:
yep that is the exact workflow i have followed on two devices with two knomis and it doesnt work for me. the dots just keep going