bigtreetech/PandaKnomi #22: Panda Knomi wrong Errors

krebsle, 2025-04-11:
The following error messages on the Panda Knomi are not actually errors and should not be displayed as such:

- Printer: Bambu Lab X1c
  Inspection of the first layer (Knomi displays it as an error until the inspection is complete)
 
- If a spool  in the AMS is empty and is automatically switched to another one, an error message will remain on the Knomi.

Please Fix this errors in the next Release

bigtreetech, 2025-04-11:
Thanks so much for reporting here. We will get these included in the firmware pipeline.

krebsle, 2025-04-11:
[image]

krebsle, 2025-04-16:
Another error is that if the localization marker was not found and the print is continued manually, the Knomi stops on an error and does not return to the normal view

FernandoGOT, 2025-04-21:
I have the same problem, if I get an error, like bad first layer or a warning like the end of a filament and changed to the second, I get stuck on this error screen until the end of my printing, even if I already fixed the problem

homeveg, 2025-06-05:
For A1mini as well wrong error is reported - In the system the filament was running out in one of the AMS slot, but Knomi displays HMS_0300-0100-0001-000A: The heatbed temperature control is abnormal, the AC board may be broken.

daneyuleb, 2025-09-14:
It's showing me a Printer error has occurred. The scan code is for the old "1st order mechanical resonance mode of x axis is low".  Once the error occurs, it stays for the whole print -- very very annoying for a long print to lose the status like this.  That's a known bogus error, and not showing anywhere but the knomi screen.  Can you please remove it? Or at the least, let us clear errors on the web ui??

xperiments, 2025-09-15:
@bigtreetech added access to all users in this issue