bigtreetech/PandaKnomi #23: Panda Knomi only shows Heating Bed during printing

hungnguyen7, 2025-04-12:
I am facing the issue that Panda Knomi just shows the Heating Bed during printing but other statuses such as Nozzle Heating, Filament Cut, etc. still work perfectly.

bigtreetech, 2025-04-12:
Have you updated to the latest firmware?

hungnguyen7, 2025-04-12:
Updated. But the problem is still not fixed. My printer is Bambu Lab A1.

N3tYaroz3, 2025-05-06:
> Have you updated to the latest firmware?

> Updated. But the problem is still not fixed. My printer is Bambu Lab A1.

Hello! I noticed that this problem occurs when I use the plate at 0 degrees. By setting a low temperature, the panda knomi will remain in this mode all the time. I hope I was helpful in trying to solve the problem.
I updated the firmware obviously.

hungnguyen7, 2025-05-07:
> Hello! I noticed that this problem occurs when I use the plate at 0 degrees. By setting a low temperature, the panda knomi will remain in this mode all the time. I hope I was helpful in trying to [...]

Yep, for me either but I no

N3tYaroz3, 2025-08-02:
Please, consider to resolve this issue! 

[image]

Jarod2801, 2025-08-08:
I also have this issue when using a 0° plate.
I haven`t updated my FW yet as the Knomi just arrivede.
But looks like the bug still exists as @N3tYaroz3 reports

kriscastnercis-sudo, 2025-08-29:
My A1 mini is also showing the same thing when using a low temp Bed setting. My A1 printing right next to it is printing PLA with a bed temp of 45c and my Knomi is working normally, but my mini printing with a bed temp of 38c is stuck showing the bed heating gif.

N3tYaroz3, 2025-08-29:
> My A1 mini is also showing the same thing when using a low temp Bed setting. My A1 printing right next to it is printing PLA with a bed temp of 45c and my Knomi is working normally, but my mini printing with a bed temp of 38c is stuck showing the bed heating gif.

Hello! The new firmware that fixes this issue is currently being tested in beta. I saw it today on Reddit.