bigtreetech/PandaKnomi #27: Errors and warnings stop all screen animation

austinwbest, 2025-05-11:
With new bambu firmware there isnt a way to clear warnings/errors and now when one happens that is all the screen will show. Add a way to ignore those with the screen animation! There is absolutely no need for this when the LCD displays them, bambu handy displays them, bambu studio displays them. 

This is supposed to be an addon for animations but it is absolutely useless when the only thing that shows up is stupid QR codes

homeveg, 2025-06-05:
Same problem to me. Even updating the firmware doesn't solve the problem of QR code on the screen
Very annoying...

austinwbest, 2025-06-05:
I have an open support ticket with Bambu as well to try and get them to add back a clear button or something. It doesnt seem this project really has much activity in terms of support or help unfortunately.

Definetly regret the purchase of Knomi's as it has been little to no communication from the start (even when it was just order/shipping problems)

bigtreetech, 2025-06-05:
There is active development ongoing for the next firmware version at the moment and this issue is being worked on.

xperiments, 2025-08-07:
@austinwbest @homeveg I will take care of this issue, can you please comment in more depth?

Continues to be a problem?
Can you specify exact versions (SW/HW) so we can target our tests in the correct direction?
Any feedback from BBL ( Do you have a link to the BBL issue you reported)?

Thanks in advance

homeveg, 2025-08-07:
Hi @xperiments, 
thanks that you've srated working on the problem

> Continues to be a problem?
> Can you specify exact versions (SW/HW) so we can target our tests in the correct direction?

I still have a Knomi IMG/Firmware version V1.0.3.1, A1mini 01.05.00.00 with AMSLite 00.00.07.94. 
And I still have described problem - A1mini complains about filamet run-out detection, change the spool and happily continue printing. Unfortunately Knomi, after the error occur, constantly shows QR code of absolutely unrelevant error message, which dose not dissapear until the very end of the print.

Going to update it today and will report it back to you, if the problem persists

Best

xperiments, 2025-08-08:
Hi @austinwbest @homeveg,

We’ve been testing this internally but haven’t been able to reproduce the error.

**Next steps:**

- We’ll release an update that allows users to disable HMS notifications as a short-term workaround.
- We’ll try to arrange a Discord meeting to see the issue live, possibly by accessing the Knomi logs.

homeveg, 2025-08-09:
I've updated the Knomi firmware to the latest version - doesn't change the way it handles the problem.
I let the printer run and change filament, and now again a flashing QR code leading to the wrong help page.
Please see attached photos.

[image]
[image]
[image]

homeveg, 2025-08-09:
> Note that this QR code is not taking you to the wrong HMS message. It is just taking you to the general HMS landing page and it happens to be that the top message on that page is the one that you have shown

In this case, it makes things even worse and more useless. The only reason which I can understand, why at all to show a QR code, which would bloack forever all animation is the possibility for user to go directly to a relevant page and not to the generic HMS landing page.
So please, either make it usefull and let us decide to use it or fully switch of, or just remove the feature.

austinwbest, 2025-08-26:
So i updated one of them & i see the new setting. It wiped out all my custom gifs which is **absolutely not cool* as i dont even know if i still have them all saved to re-add them all again.

xperiments, 2025-08-26:
@austinwbest Thank you again for testing.

The “unknown” inconsistencies appear to be caused by your phone caching the web app, which explains why it worked correctly after the power cycle.

Regarding the “replaced” GIFs: we have added a new GIF for the ERROR status. Because of this change, a full recovery of the original GIFs is required. If you generate a new IMG file, it will already include the new GIF, ensuring everything is consistent going forward.

Finally, we would appreciate your feedback on the original error — is it now resolved on your side after the introduction of the OFF / QR / GIF option?

Thanks!!

bigtreetech, 2025-08-26:
> So i updated one of them & i see the new setting. It wiped out all my custom gifs which is **absolutely not cool** as i dont even know if i still have them all saved to re-add them all again.
> 
> As a side note, if this is expected behavior during an update i will not update this again. That is rediculous.

The firmware is still in beta. By definition, there could be unexpected behaviour with the GIFs since we are revamping a few of them given the new error handling. For all others interested in testing this beta, please take note and back up your GIFs.

homeveg, 2025-08-28:
I've updated the Knomi firmware/images only from the third or forth try. 
Not sure what helped exactly, as long as I did power cycing as well as the cleaning browser cash on my PC.
Regarding the main issue - I've set the Knomi Error Display Mode to a "Error Display GIF". 
After filament was automatically loaded from the AMS, I've got the progress status GIF as before.
Thanks for fixingf it!

xperiments, 2025-08-28:
> I've updated the Knomi firmware/images only from the third or forth try. Not sure what helped exactly, as long as I did power cycing as well as the cleaning browser cash on my PC. Regarding the main [...]

Glad to hear it’s working now! Thanks for reporting the cache-related issue — I’ll open a new issue to track adding a non-caching option on the web side to avoid similar problems in the future.

xperiments, 2025-08-29:
> > Glad to hear it’s working now!
> 
> Since I didn't notice when the filament ran out last time, I checked it again by cutting the filament before the AMS inlet — when the filament was completely depleted, and Knomi was in the mode mentioned above (‘Error Display GIF’), no error messages or GIF files related to the error were displayed on the screen. It just cycles through the various stages associated with filament loading. For this particular error, I consider this normal, but perhaps this is not the expected behaviour? Or did I choose the wrong settings?

Hello,

This is expected — the current system will only show “critical” Print errors that actually block the process.

The previous error system triggered on all kinds of errors. While some disappeared once resolved, others remained as historical entries, which eventually caused the problem reported in this issue.