bigtreetech/PandaKnomi #29: Knomi Won't connect to WiFi

JustinPalmer-AI, 2025-06-07:
I received a Knomi today and despite mine and my wifes best efforts we cannot get it to connect to WiFi. I have updated to the most recent firmware Via USBC v1.0.3.1  The UI on my phone displays "Wi-Fi connection issue(201), reconnecting..." and it never connects.

Any help is much appreciated.

Thank you!

donnyad, 2025-09-04:
I had an issue as well using a EERO Mesh router. i had to edit some router settings to allow it to work. First i reserved the ip for the panda knomi. forcing it to keep its same ip making future setups easier. i also reserved my printers ip as well. then i had issues getting the knomi to see my printer. i had to disable SQM (Smart Queue Management and Thread. those 2 settings make it hard for devices to communicate on the network. another thing is there is a  way to temporarily disable the 5g internet so it forces all devices to use 2.4. most 5g devices can utilize 2.4 but not vice versa. eero routers and many other ones combine the internet bands now leaving it to the device to differentiate which creates issues with a lot of devices. lastly i created a "group" with my panda knomi, printer, wifi cam, and a few other printer upgrades to ensure eeveyrthing stays organized.

JustinPalmer-AI, 2026-11-14:
@BTTTony98 I haven't Unfortunately I completely forgot about this problem and bought the panda breath so now I have a 100-dollar paperweight in my printer