bigtreetech/PandaKnomi #31: Knomi update img - then unresponsive

dexteriade, 2025-06-11:
Hi, new panda knomi installed on X1C. 
I have connected the knomi to the printer - succesfully after modifying AP setting not to be allways on. 
After... i did a possible booboo. I thought to follow the recomandation to have the latest updates for the knomi. I intered the link from the wiki, downloaded locally the V1.0.3.1 folder containing the .img and .bin files. I did not read if there is a certain order for update. 
I accesed knomi.local and.... updated the img (in was first on the page) to v1.0.3.1.img. File selected ok, update ok... on the knomi display there was an error message that the file is not ok and a qrcode... i did not make a picture of the error message. just... hit restart to the printer. 
Now knomi is uresponsive. does not turn on. knomi.local is unavailable... 
did i brick it?

bigtreetech, 2025-06-11:
You would need to post a pic of the error message. Also, you can never brick it because you can always force the firmware down via the USB-C port. There is a wiki section explaining how to do this.

dexteriade, 2025-06-11:
[image]
[image]

closed 2025-06-13 by dexteriade