bigtreetech/PandaKnomi #32: Panda Knomi - not updating to V1.0.3.1

dexteriade, 2025-06-13:
Hi. New panda knomi, trying to update via pandaknomi.local. 
I downloaded the firmware from github and tried directly v1.0.3.1. i receive an error message. 
I then downloaded V1.0.3 and i was able to update to 1.0.3. 
I tried again v1.0.3.1 and the same error message. 
Sugestions on what to try next?

it is mounted on a X1C with 2 AMS units.

BTTTony98, 2025-06-13:
https://bttwiki.com/PandaKnomi.html#ota
You need to select both img (v3.0.1) and bin (v3.0.1) files

dexteriade, 2025-06-13:
You cant select both. there are individual fields and selecting a file automatically triggers the update. And the wiki mentiones only the  bin file. Anyway i said i will try... - i have updated the IMG to 1.0.3.1 and the message was "Successful" but at the version then it shown "unknown". And after i tried to upgrade de bin to 1.0.3.1 and the same "update error" message". Thus i tried reverted the img to the 1.0.3 in order not to have missmatches between versions but... no more communication to the knomi. Now i have to make a hard reset :))) offf.

BTTTony98, 2025-06-16:
Or you can use USB-C to update the firmware 
https://bttwiki.com/PandaKnomi.html#flash_download_tool-type-c

closed 2025-10-30 by xperiments